elixir-circuits/circuits_gpio #21: Rename set_edge_mode

fhunleth, 2018-11-12:
`set_edge_mode/3` is feeling like it may not be the best name any more and I'm having mixed feelings about having both a keyword list and a default edge mode argument. What do others think?

To get things started, how about:

`set_interrupts(ref, enabled, options \\ [])`

where `enabled` is `true` or `false`. Options are:

* `edge` - `rising`, `falling`, or `both`. The default is `both`
* `suppress_glitches` - `true` or `false`. The default is `true`.
* `receiver` - pid/name of where to send notification messages. The default is `self()`

mdsebald, 2018-11-13:
Maybe.  My thinking was that one purpose of `set_edge_mode()` or 'set_interrupt()`  is to set the edge dev virtual file to one of those 4 valid values, and I didn't think of that as an option, but rather a main parameter.  Adding level-triggered selections, feels like we're getting into the area of options.   I still agree with changing the function name.  I don't have a strong feeling about the name of the of the parameter, but I like the one-to-one match of the value of that parameter, and the set of allowed values for the edge file.

fhunleth, 2018-11-13:
Yeah, I agree with options vs. mandatory. Currently the trigger mode is an optional parameter. I suspect that we're in agreement that that should change. Are we closing in on:

`set_interrupts(ref, trigger, options \\ [])`

I'm fine with keeping trigger as `none`, `rising`, `falling` and `both` like it currently is.

commit a636d227 (2019-01-02): Rename set_edge_mode to set_interrupts

This fixes #21. The desire is to improve the name of the function to
make it more obvious what it does (i.e. configure the sending of
messages when the line changes state). See the issue for the discussion.
PR #26 (2019-01-02): Rename set_edge_mode to set_interrupts
This fixes #21. The desire is to improve the name of the function to
make it more obvious what it does (i.e. configure the sending of
messages when the line changes state). See the issue for the discussion.
commit e1ea1a00 (2019-01-02): Rename set_edge_mode to set_interrupts

This fixes #21. The desire is to improve the name of the function to
make it more obvious what it does (i.e. configure the sending of
messages when the line changes state). See the issue for the discussion.
closed 2019-01-02 by fhunleth