elixir-circuits/circuits_gpio #22: Look up registered names when sending notifications

fhunleth, 2018-11-12:
Currently, registered process names are supported, but the name-to-pid translation happens when interrupts are enabled. To handle the case where the registered process dies and comes back, the lookup really should happen on the send call. This involves changes to the NIF.

mdsebald, 2018-11-13:
OK. I see `erl_nif` has an `enif_whereis_pid()` function.  So the only downside is adding logic to determine if that should be called, on each interrupt.  Probably not a big deal.

fhunleth, 2023-12-30:
We're not addressing this in v2. 

This may not be an issue since the idiom for using Circuits.GPIO for inputs is to open the GPIO in the process handling the notifications. That makes it easier to give the guarantees to the receiving process that the GPIO is set up correctly and hasn't crashed or anything like that.

closed 2023-12-30 by fhunleth