elixir-circuits/circuits_gpio #49: [question] Library expansion for common use cases

mroach, 2019-03-09:
I'm opening this issue to discuss what the direction of this library is intended to be. First off, it's awesome having a GPIO library in Elixir that works nicely on a Raspberry Pi. Now Python and C don't have to be the only options.

Question is, is there motivation to expand the library to cover common use cases for GPIO? One example being buttons. I spend a bit time coming up with a way to handle buttons with each one being a `GenServer`. [Source](https://github.com/mroach/rpi_tinker/tree/master/lib/tinker).

TLDR usage:

```elixir
  # part of application.ex
  def start(_type, _args) do
    children = [Buttons.PlayPause, Buttons.PreviousTrack, Buttons.NextTrack, Buttons.Shuffle]
    Supervisor.start_link(children, strategy: :one_for_one)
  end
```

```elixir
defmodule Tinker.Buttons.PlayPause do
  use Tinker.Button, pin_id: 26

  def pressed(_event), do: IO.puts "Play!"

  def released(%{duration: duration}) do
    Logger.debug("Play was depressed for #{duration}ms")
  end
end
```

```shell
Play!

12:54:35.724 [debug] Play was depressed for 1175ms

Next!

12:45:09.779 [debug] Next was depressed for 120ms
```

Does it make sense to include something like this into `circuits_gpio`? If so, I'd be happy to package this up into a PR, add testing, documentation. I think making common use cases easy would be a win for adoption.

fhunleth, 2019-03-09:
I definitely agree that there's a need for a generic button and a LED library. The button library would have to support polarity selection and debouncing, so it's not trivial. I really like the feature set offered by gpio_zero at https://gpiozero.readthedocs.io/en/stable/api_input.html#button. 

`circuits_gpio` is getting close to 1.0. The API hasn't changed much in the past six months. Figuring out a good button library may take a while and then there's a discussion on what additional things get supported. It feels a lot better to me to keep `circuits_gpio` focused on the low level interface since that part is already a bit complicated and time consuming for us to maintain. 

The circuits home page points people to libraries that build off `circuits_gpio`, so a button library, for example, would be somewhat discoverable if you were to make one and publish to hex.pm. If you're interested in making a reasonably full-featured button library like gpio_zero does, naming it `circuits_button` and including it under the Elixir Circuits projects could make a lot of sense. I think that you should start that up and give it a few iterations before we go down that path, though.

mroach, 2019-03-10:
Makes total sense to me to keep `circuits_gpio` lean and focused on the low level. My second idea/option was going to be expanded documentation or examples of how to implement common things like LEDs, buttons, and buzzers. For example how to capture the interrupts for button presses. Where would the best place for that be? Application docs that end up on hex.pm, wiki, separate project?

I'll work on some iterations for a button library then and figure out how to have some features from gpiozero done the Elixir way.

mroach, 2019-03-10:
Thanks for the feedback on the idea. I'll close this as I just need a place to ask :)

closed 2019-03-10 by mroach