elixir-circuits/circuits_gpio #53: Renaming :pull_mode options

jjcarstens, 2019-03-19:
I think there is a slight mismatch in naming for the `:pull_mode` options in `open/3` and `set_pull_mode`.

I.e. the option is `:pull_mode` which is snake case, but the supported atoms are `:pulldown` and `:pullup` which are compressed.

What do you think about combining to make them all either snake case (`:pull_mode`, `:pull_up`, `:pull_down`) or compress `:pullmode`, `:pullup`, `:pulldown`)? Or, we could just support both compressed and snake case pull mode options?

fhunleth, 2019-03-20:
I’ve always thought of pullup as one word, but I see that the official spelling is “pull-up”. I assume that in snake case that’s pull_up like you’re proposing? If so, then I’m for changing. Thanks for noticing this before 1.0!

fhunleth, 2019-03-27:
@jjcarstens I'm actually not sure on the rule for hyphenated words with Elixir. Are the hyphens removed or turned into underscores? Are there examples in the main Elixir code? Anyway, let's circle around on this and PR it if necessary.

fhunleth, 2019-04-01:
From The Art of Electronics:

> In logic circuits they act as bus and line terminators and as “pullup” and “pull-down” resisters.

I'm going to leave the spelling as is. I don't think there's any winning here. We could support both spellings, but I find that leads to more confusion.

closed 2019-04-01 by fhunleth