elixir-circuits/circuits_gpio #54: Add ASCII art help for pinout numbering

fhunleth, 2019-03-27:
The Raspberry Pi has a really great tool called `pinout` to help users figure out which pin is which. It would be really useful if something similar were available for `Circuits.GPIO`. Since `Circuits.GPIO` is cross platform, we have to be careful on how we implement this so I'm looking for ideas. 

Here's what `pinout` looks like:

[image]

Really, just the pins are needed. Creating a separate project (`Circuits.Pinout`?) might be the right thing to do as well.

mattludwigs, 2019-03-27:
This might be over-engineering a bit, but could this be in a separate project, but make it a mix task where either we read in the `MIX_TARGET` value or let the user pass in the target or maybe even have it be interactive?

I like this idea, so I am just split ballin' an idea that you can shoot down :), but it would be awesome to provide great tooling around circuits to help development, debugging, and learning.

fhunleth, 2019-03-27:
I'm leaning towards separate project as well.

Regarding the implementation, `MIX_TARGET` doesn't provide any board information since you can set it to anything you want. You can also be running on Raspbian, for example, and `MIX_TARGET` wouldn't even be set. There are heuristics that can be done to identify where you're running. The Raspberry Pis boards can be identified by string parsing `/proc/cpuinfo`. Some investigation is needed for other boards.

mattludwigs, 2019-03-27:
👍 that makes sense

batate, 2019-03-27:
Forgive my ignorance. I am very new at nerves. Is this: 

- a registry 
- that we dump the GPIO pin numbers and refs  as they are opened (and written)
- and an output tool that maps over them, showing the values of the open pins?

fhunleth, 2019-03-27:
Yes, maybe, I'm not sure. I opened this issue to understand better what would be helpful. I think that I jumped to solution that I was familiar with rather than listening fully.

The Python `pinout` tool identifies which Raspberry Pi it's running on and then outputs a ASCII art picture of the pin assignments and other information. Unless it's been updated since I last looked, that was it.

Making an output tool that maps over the opened pins and shows information (input, output, value, etc) sounds very useful too. I like that.

mattludwigs, 2019-03-27:
I like that too, a lot.

mattludwigs, 2019-03-27:
So it sounds like we have some features here:

1. Pint the pinout inspired by the python `pinout`
1. Get data about opened pins

Do you think both those belong is a separate tool together?

I can see the `pinout` tool being a new repo, but maybe the open GPIO data being in this library?

Just asking some questions to get a better idea of what people are thinking.

ChristopheBelpaire, 2020-07-10:
I'm trying to debug it and it seems to block at the load of the nif : 
```:erlang.load_nif(to_charlist(nif_binary), 0)```

fhunleth, 2022-12-31:
This can be closed now that there's a pinout library in Elixir. See https://github.com/cavocado/pinout.

closed 2022-12-31 by fhunleth