elixir-circuits/circuits_gpio #57: Stub mode does not turn on

corka149, 2019-06-24:
Hello there

I tried to create some tests for my little nerves project in combination with circuits_gpio but I found an issue with the stub mode while using circuits_gpio as dependency.

## Expected Behavior
When running "mix test", GPIO.info should return a tuple with "{name: :stub, pins_open: 0}.

## Current Behavior
GPIO.info returns actually "{name: sysfs, pins_open: 0}".

## Steps to Reproduce
Example: https://github.com/corka149/alerter/blob/master/test/alerter_test.exs

Steps to reproduce
1. Create a new mix project.
1. Add circuits_gpio as dependency
1. Copy test from "info returns a map" from [circuits_gpio/test/circuits_gpio_test.exs](https://github.com/elixir-circuits/circuits_gpio/blob/master/test/circuits_gpio_test.exs)
1. Run "mix test"

## Note
In addition I cloned circuits_gpio and run the mix test task. The named test succeed.

fhunleth, 2019-06-25:
Unfortunately, there's no way to manually turn the stub on. There should be. If you'd like to implement, feel free to send a PR or one of us will get around to it in the coming weeks.

corka149, 2019-06-25:
Thanks for your reply. 

For anyone who is coming round this issue:
The hal_stub.c is not compiling. When I set "MIX_ENV=test" and "CROSSCOMPILE=true" it prints out that gcc, make or erlang-dev is missing. But they exist on my system. If I manipulate the Makefile so that it will always compile the hal_stub.c, then it works via "mix test".

I will try to figure out why this is happening asap.

corka149, 2019-06-29:
A note was add to the README.md how it is possible to use the stub.

closed 2019-06-29 by corka149