elixir-circuits/circuits_gpio #62: Raspberry Pi 4 model B :: GPIO Pullup modes

jsimmonds2, 2019-09-09:
Have a simple app which uses GPIO 12 as `input` connected to an open-drain output on a DS3231 RTC.
Have tried setting **`:pull_mode`** in the **`GPIO.open`** options according to the documentation
but cannot get a **`:falling`** edge interrupt.
```
    {:ok, gpio_pin_12} = GPIO.open 12, :input, [{:pull_mode, :pullup}]    # A 1 Hz SQW or Alarm 1 interrupt from the DS3231
```
```
    GPIO.set_interrupts gpio_pin_12, :falling
```
As I don't have a scope, I'm going to try a 47k external pullup and see if that helps . . now off to order that resistor :)

mdsebald, 2019-09-09:
Are you getting the initial message on the GPIO.set_interrupts call?  

Then, if you disconnect the RTC and just manually short GPIO 12 to ground, are you still not getting the interrupt message?

jsimmonds2, 2019-09-10:
> Are you getting the initial message on the GPIO.set_interrupts call?

Only when GPIO 12 is tied to GND and **not** when tied to +3V3.

> Then, if you disconnect the RTC and just manually short GPIO 12 to ground, are you still not getting the interrupt message?

That is so.

Sure looks like there's a **`:pulldown`** on that GPIO.
I can take that lead from GND and dab it onto the +3V3 rail and I get **`:falling`** edge interrupts, with pin value 0.

mdsebald, 2019-09-10:
I don't own an RPi 4.  I hope nothing has changed in the GPIOs.  I'll take a closer look when I have more time this evening (US Central TZ).

mdsebald, 2019-09-11:
Sorry, unable to verify if issue exists on my setup or not.  Will try again tomorrow evening.  I have ordered an RPi 4.  I will test with that, also.

jsimmonds2, 2019-09-11:
Wow !  Thanks for digging deeper . . I was actually thinking about offering to send you an RPi4 to test with.

mdsebald, 2019-09-12:
First I verified that the internal pull-up and pull-down resistors are still working as expected with Circuits.GPIO on the RPi3.  I tried the same on an RPi4, and indeed, it appears the internal pull-up resistors (I only tried on GPIOs 12 and 23 so far) are not being switched in via the pullmode pullup option.  I don't know what is wrong and not sure what it will take to fix it yet.  Hopefully, you can get by with an external pull-up resistor for now.  Just in case the pullmode option is doing something undefined on the RPi4, I would also take out that option, when opening an input pin.  Thanks for bringing this to our attention.

jsimmonds2, 2019-09-13:
> FYI: I found this code too
> [https://github.com/raspberrypi/linux/blob/e2a6ce9ff7c94cce5c1d40d2c79a2cd0498d5f88/drivers/pinctrl/bcm/pinctrl-bcm2835.c#L933-L954](url)

Hi Mark.  Could you edit your link to put the URL inside the parentheses rather than the brackets.  :)

mdsebald, 2019-09-13:
Done.  Thanks!

mdsebald, 2019-09-17:
I updated src/hal_rpi.c  to include RPi4 pull modes code (branch: implement_rpi4_pullmodes).  Tested pull-up and pull-down modes on one GPIO pin (12) on the RPi4 and it worked OK.  Still need to do more testing including regression tests on other Pi models, before submitting PR.

fhunleth, 2019-09-20:
Fixed in #63. Thanks @mdsebald.

closed 2019-09-20 by fhunleth

jsimmonds2, 2019-09-21:
Appreciate your fixes, @mdsebald 👍
I'm now "one happy end-user",  getting the ( Active Low ) Alarm interrupts from the DS3231 RTC exactly as described 🙂  . . _**without**_ any external pullup resistor.

mdsebald, 2019-09-21:
Thanks to @fhunleth for pushing it over the finish line too, Glad we were able to resolve this issue.