elixir-circuits/circuits_gpio #78: Release doesn't boot when using circuits_gpio with releases.exs

ChristopheBelpaire, 2020-07-10:
Hello,
I have a strange problem with release, the configuration releases.exs and circuits_gpio.
When the file releases.exs is present in my project, the release will never start.
If the releases.exs file is not present everything works fine.
If the dependency of circuits_gpio is not present, everything works fine also.

Step to reproducing the problem :

```mix new release_test```
add the dependency circuit_gpio to your mix file
```mix deps.get```
create an config/releases.exs in the project
```mix release```
```_build/dev/rel/release_test/bin/release_test start_iex```
```
pi@raspberrypi:~/release_test $ _build/dev/rel/release_test/bin/release_test start_iex
Erlang/OTP 22 [erts-10.5.4] [source] [smp:4:4] [ds:4:4:10] [async-threads:1] [hipe]
````

I'm on a rasperry pi 3 with :
Erlang/OTP 22 [erts-10.5.4] [source] [smp:4:4] [ds:4:4:10] [async-threads:1] [hipe]
Elixir 1.10.3 (compiled with Erlang/OTP 21)

Any idea where it could come from ?

Thanks in advance!

ChristopheBelpaire, 2020-07-10:
I'm trying to debug this and it seems to block at the load of the nif :
```:erlang.load_nif(to_charlist(nif_binary), 0)```

mdsebald, 2020-07-10:
Could this be a permissions issue?  Since the code is manipulating GPIO registers, it needs root access.  Also, users of the Elixir  Nerves project often use the Circuits libraries, so posting questions to the Elixir Nerves Slack channel or the Elixir Forum Nerves category, will give your question a much wider audience.  Hope this helps.

ChristopheBelpaire, 2020-07-11:
I don’t think so, it also blocks with sudo.
I’ll try to debug the nif to see where it blocks.

ChristopheBelpaire, 2020-07-11:
Here are the logs of the circuit_gpio c parts when I launch the release :
```load
gpio_poller_thread started
unload```

ChristopheBelpaire, 2020-07-11:
If I remove the "unload" function, It fix my problem.
But I don't know enough about Nif system to understand if it is a good idea ;)
https://github.com/ChristopheBelpaire/circuits_gpio/commit/1d9cf6514b6c8fce54a4712f0b2b1cb25110cdc1

fhunleth, 2020-07-12:
Thanks for debugging this!

I don't know what the ramifications of removing `unload` are either.

I think that you're the first person to report a call to `unload` for `circuits_gpio`. Looking at the `unload` code, it seems incomplete and that may be enough to cause a crash later on. 

I think we should stage the fix. Could you make a PR with your removal of the `unload` registration and also remove the `unload()` function? We should make an official release with that change so that the `releases.exs` use case works. Since I don't believe anyone else is calling `unload`, this seems like a low risk change. Properly fixing `unload` seems like it will take some thought.

Also, as a side note, Elixir 1.11 is going to introduce `runtime.exs` and encourage people to switch from `releases.exs`: https://github.com/elixir-lang/elixir/blob/12001c6c2832ffe55992f93c96be1b55b057853c/CHANGELOG.md#configruntimeexs-and-mix-appconfig. I haven't tried `runtime.exs` yet, but I wonder if it will have a similar issue.

ChristopheBelpaire, 2020-07-13:
No problem,
here is the pull request : https://github.com/elixir-circuits/circuits_gpio/pull/79
I still don't understand the link between the unload and the releases.exs file, but it seems to fix the problem :)

fhunleth, 2020-07-24:
Sorry for the delay - haven't forgotten.. just busy.

PR #91 (2021-07-02): Terminate interrupt thread on NIF unload
This fixes an issue where the NIF could not be unloaded due to the
interrupt monitoring thread never stopping. This would manifest itself
in `:init.stop` taking a long time or it breaking `runtime.exs` due to a
change that requires the VM to be restarted.

Fixes #78
closed 2021-07-02 by fhunleth