elixir-circuits/circuits_gpio #84: aarch64 on RPi4B breaks GPIO.open

jsimmonds2, 2020-11-27:
First things ~  Thanks very much for creating a 64-bit ARM Linux Buildroot version !!

I'd love to be tooled up and savvy enough to fix this one quickly but I'm not . . hence this Issue 🙂 

This statement used to work **:ok** on 32-bit ( **nerves_system_rpi4** `v1.13.0` ) but breaks on `v1.13.1`

```elixir
iex(8)> {:ok, gpio_pin_12} = Circuits.GPIO.open 12, :input, [{:pull_mode, :pullup}]
** (MatchError) no match of right hand side value: {:error, :error_setting_pull_mode}
```

fhunleth, 2020-11-27:
Wow, that makes a lot of sense that pullups broke on the switch to 64-bits. Thanks for the quick heads up.

I assume that the memory layout is different on 64-bits and we just need to update the register offsets in C. I'll see if I can find the new address later today.

mdsebald, 2020-11-29:
I built circuits_quickstart for the rpi4 to test a fix when it became available, but was not able to duplicate the problem in the first place.

```Elixir
iex(18)> {:ok, gpio_pin_12} = Circuits.GPIO.open 12, :input, [{:pull_mode, :pullup}]
{:ok, #Reference<0.3936564783.537526284.69761>}
```

Build environment:
```
mark@MARK-LAPTOP-1:~/nerves/elixir_circuits/circuits_quickstart$ asdf current
elixir         1.11.2-otp-23 (set by /home/mark/.tool-versions)
erlang         23.0.1   (set by /home/mark/.tool-versions)
```

I think I'm using nerves_system_rpi4 version 1.13.1 and nerves_toolchain_aarch64_unknown_linux_gnu

[mix.lock.txt](https://github.com/elixir-circuits/circuits_gpio/files/5612525/mix.lock.txt)

I'm guessing there's something different about my environment.  If there is some other way I can duplicate the problem, please let me know. Thanks.

fhunleth, 2020-11-29:
@mdsebald Your `mix.lock` file looks right.

@jsimmonds2 Is it possible that `circuits_gpio` wasn't rebuilt after you updated `nerves_system_rpi4`? If so, I think that it's possible to add a check to the Nerves tooling for this issue, but I want to confirm first.

jsimmonds2, 2020-11-29:
@fhunleth I believe that `circuits_gpio` v0.4.6 **_is_** being built during `mix firmware`. Issue is 100% repeatable for me.
My major environment difference w.r.t. @mdsebald seems to be Elixir and Erlang versions. I'm on Fedora 32 Workstation (virtual)

Transcript of my latest build below . .  Thanks !

```text
[jhs@f32v-on-comet fb2]$ asdf current
elixir          1.10.4-otp-23   /home/jhs/src/jsimmonds2/RasPi3BPlus/Blinkers/Nerves/fb2/.tool-versions
erlang          23.1.1          /home/jhs/src/jsimmonds2/RasPi3BPlus/Blinkers/Nerves/fb2/.tool-versions
[jhs@f32v-on-comet fb2]$ mix clean
==> nerves
==> fb2

Nerves environment
  MIX_TARGET:   rpi4
  MIX_ENV:      prod

[jhs@f32v-on-comet fb2]$ mix deps.clean --all
* Cleaning certifi
* Cleaning circuits_gpio
* Cleaning circuits_i2c
* Cleaning circuits_uart
* Cleaning dns
* Cleaning elixir_make
* Cleaning gen_state_machine
* Cleaning mdns_lite
* Cleaning muontrap
* Cleaning nerves
* Cleaning nerves_pack
* Cleaning nerves_runtime
* Cleaning nerves_ssh
* Cleaning nerves_system_br
* Cleaning nerves_system_rpi4
* Cleaning nerves_time
* Cleaning nerves_time_rtc_ds3231
* Cleaning nerves_toolchain_aarch64_unknown_linux_gnu
* Cleaning nerves_toolchain_ctng
* Cleaning one_dhcpd
* Cleaning parse_trans
* Cleaning ring_logger
* Cleaning shoehorn
* Cleaning socket
* Cleaning ssh_subsystem_fwup
* Cleaning system_registry
* Cleaning tortoise
* Cleaning uboot_env
* Cleaning vintage_net
* Cleaning vintage_net_direct
* Cleaning vintage_net_ethernet
* Cleaning vintage_net_wifi
[jhs@f32v-on-comet fb2]$ mix deps.get
Resolving Hex dependencies...
Dependency resolution completed:
Unchanged:
  certifi 2.5.2
  circuits_gpio 0.4.6
  circuits_i2c 0.3.7
  circuits_uart 1.4.2
  dns 2.2.0
  elixir_make 0.6.1
  gen_state_machine 2.1.0
  mdns_lite 0.6.6
  muontrap 0.6.0
  nerves 1.7.0
  nerves_pack 0.4.1
  nerves_runtime 0.11.3
  nerves_ssh 0.2.1
  nerves_system_br 1.13.4
  nerves_system_rpi4 1.13.1
  nerves_time 0.4.2
[... 290 lines cut ...]
Number of device nodes 0
Number of fifo nodes 0
Number of socket nodes 0
Number of directories 259
Number of ids (unique uids + gids) 3
Number of uids 3
        root (0)
        unknown (33)
        jhs (1000)
Number of gids 3
        root (0)
        tape (33)
        jhs (1000)
Building /home/jhs/src/jsimmonds2/RasPi3BPlus/Blinkers/Nerves/fb2/_build/rpi4/rpi4_prod/nerves/images/fb2.fw...
[jhs@f32v-on-comet fb2]$
[jhs@f32v-on-comet fb2]$ # -------------------
[jhs@f32v-on-comet fb2]$
[jhs@f32v-on-comet fb2]$ cat _build/rpi4/rpi4_prod/nerves/images/fb2.fw | ssh 192.168.0.100 -s fwup
fwup: Upgrading partition A
100% [====================================] 29.59 MB in / 36.73 MB out
Success!
Elapsed time: 5.205 s
[jhs@f32v-on-comet fb2]$

```

fhunleth, 2020-11-30:
I'm able to reproduce. @mdsebald I'm going to take a quick look to see if I can narrow the issue down a little. I'm testing with the `circuits_quickstart` `main` branch.

fhunleth, 2020-11-30:
Found it. The RPi pull up/down code depends on bcm_host.h. bcm_host.h is supplied by RPi userland. Buildroot has a rule that doesn't allow RPi userland to be built for 64-bit ARM. So, it turned it off to be helpful. I need to look around for options - not sure if the right answer is to patch BR to build the bcm_host part or whether `circuits_gpio` should stop using bcm_host.

mdsebald, 2020-11-30:
FYI I started from scratch,  and I am able to reproduce the problem now.

fhunleth, 2020-11-30:
Good news: I saw pullups work again with https://github.com/nerves-project/nerves_system_br/pull/500. 

Bad news: This is a patch to Buildroot, so getting this to propagate to a fixed RPi4 system will take a little time. It's not complicated - just time consuming. I'll chip at it every time I get a free moment this week.

closed 2020-11-30 by fhunleth