elixir-circuits/circuits_gpio #125: PWM access

mtrudel, 2022-09-20:
Excellent folks,

Is PWM access from circuits_gpio possible, at least on RPis? It doesn't look like it's in the library, and I may take a stab at implementing support for it if nobody has any horror stories / reasons it would never work. 

Any pitfalls I'm missing before jumping in?

fhunleth, 2022-09-21:
Hi @mtrudel,

There's some history to PWM support. I avoided it with circuits_gpio since it was a bit of a mess years ago and it looked hard to maintain. That was because the Linux PWM interface was not well supported and people were writing platform-specific libraries with varying APIs and resource requirements. I don't think that's been the case for a while.

I'll be honest in that I've never used Linux's [pwm](https://elixir.bootlin.com/linux/v5.10.144/source/Documentation/ABI/testing/sysfs-class-pwm) API, but it really looks like it could be a pure Elixir library. I'd lean on keeping it separate from `circuits_gpio` for ease of maintenance and flexibility with API and dependencies. If you wanted to host the library under `elixir-circuits`, I'd definitely be onboard with that. Whatever you do, let me know since I'd like to at least link to your work so people who need PWM support can easily find it.

mtrudel, 2022-09-21:
Digging deeper, it seems that the RPi's PWM channels are each duplicated on a pair of pins (that is, if I'm looking to drive PWM on GPIO 19, that same output will appear on GPIO 13. For the narrow niche use case I'm looking at that's a deal breaker, so I'm going to explore some I2C based options. Too bad :(

closed 2022-09-21 by mtrudel