elixir-circuits/circuits_gpio #126: Stub doesn't work when setting env variables

pancho-villa, 2022-10-12:
Theoretically if you set the following environment variables, the stub should be available for testing on the host machine:

`MIX_ENV=test
MIX_TARGET=host
CROSSCOMPILE=true
CIRCUITS_MIX_ENV=test
`
But that doesn't work currently. However if you modify the Makefile you can get it to work provided you delete the deps/circuits_gpio folder from your project, re-download it, and make your modification _before compiling_ for tests or anything. I have confirmed that after the modification, it still works on the target machine like a Raspberry Pi, but I haven't confirmed there aren't other side effects yet. Here's my work around, edit the [Makefile on line #34](https://github.com/elixir-circuits/circuits_gpio/blob/498f787ad32a43fb5ef7690004ecdc6730ff38ad/Makefile#L34) to this:

`ifeq ($(CROSSCOMPILE), true)`

and modify [line #42 of the same Makefile](https://github.com/elixir-circuits/circuits_gpio/blob/498f787ad32a43fb5ef7690004ecdc6730ff38ad/Makefile#L42) to this:

`        ifeq ($(CIRCUITS_MIX_ENV), test)`

fhunleth, 2022-10-14:
Thanks for the bug report.

I'm a little confused. Why are you setting `CROSSCOMPILE=true`? `CROSSCOMPILE` shouldn't need to be set manually and it's not a true/false environment variable. Did you see this in other directions?

pancho-villa, 2022-10-14:
That's a great question. I found this [previously closed issue](https://github.com/elixir-circuits/circuits_gpio/issues/57), the bug reporter said `CROSSCOMPILE=true` was part of their workaround, but they didn't say how. 

That started my investigation wherein I noticed on line 34 that CROSSCOMPILE directive matched nothing. That's why I changed it. Now when you set that to true, it'll match and let you get to line 42, where my other change will match if you have `CIRCUITS_MIX_ENV=true` set in your environment. In my chat in slack with @jjcarstens I noticed however I set MIX_ENV variable it always evaluated to prod. He said this:

> Hm. The stubbing probably needs to be rethought. When circuits gpio is a dependency, it would always be `MIX_ENV=prod` 

Which I interpreted as MIX_ENV is set to prod _when_ circuits.gpio is a dependency. That matched with my testing as I could never get it set to anything else, hence me taking out of that filter directive on line 42 and relying on the match of CIRCUITS_MIX_ENV=test matching to the string `test` to enable the stub to be compiled.

I'm happy to elaborate more and I hope I'm not completely off my rocker.

fhunleth, 2022-10-14:
Perhaps I need an example to try and reproduce this. I just created an empty project and added `:circuits_gpio` to the deps. Here's what I see:

```
fhunleth@sprint:~/experiments/gpio_experiment$ mix deps.get
Resolving Hex dependencies...
Dependency resolution completed:
Unchanged:
  circuits_gpio 1.0.1
  elixir_make 0.6.3
* Getting circuits_gpio (Hex package)
fhunleth@sprint:~/experiments/gpio_experiment$ CIRCUITS_MIX_ENV=test mix test
==> circuits_gpio
"**** MIX_ENV set to [prod] ****"
"**** CIRCUITS_MIX_ENV set to [test] ****"
Makefile:43: Compiling stub NIF to support 'mix test'
 CC hal_stub.o
 CC nif_utils.o
 CC gpio_nif.o
 LD gpio_nif.so
Compiling 2 files (.ex)
Generated circuits_gpio app
==> gpio_experiment
Generated gpio_experiment app
..
Finished in 0.02 seconds (0.00s async, 0.02s sync)
1 doctest, 1 test, 0 failures

Randomized with seed 793023
fhunleth@sprint:~/experiments/gpio_experiment$
```

pancho-villa, 2022-10-14:
I think I found my issue, I assumed I could export my environment variables and have that work ala `export CIRCUITS_MIX_ENV=test` and several command later try mix test. It does work when I use `CIRCUITS_MIX_ENV=test mix test` all in one command tho. Am I wrong to assume the exporting of variables should work?

pancho-villa, 2022-10-14:
I can close this, it looks like it does work when you set the environment variable `CIRCUITS_MIX_ENV=test`. I can get it to fail if I don't set that variable first, and run the test, but if I remove the `deps/circuits_gpio` folder, export the variable, run `mix deps.get && mix test` it will work. It's a workflow issue on my end, thanks for looking into it.

closed 2022-10-14 by pancho-villa

fhunleth, 2022-10-15:
Sorry, I got distracted yesterday and I'm getting back to this now. I understand why Jon said that it could be improved. The workflow is too easy to mess up.

I'm guessing that you're find now since you understand the problem and can work around it. However, if you have any thoughts at improvements, feel free to PR. If not, that's ok, it's on my radar now. I'm not sure when I'll get to it, but I understand. Thanks for posting.