elixir-circuits/circuits_gpio #159: Test circuits_gpio v2.0.0-pre.* on official Nerves targets

fhunleth, 2023-12-30:
This issue tracks testing on various systems.

Two kinds of tests are useful to report:

1. Testing in apps that use Circuits.GPIO v1
2. `Circuits.GPIO.Diagnostics.report(gpio1, gpio2)` where you connect gpio1 and gpio2 together and let the `Diagnostics` module check things out.

Please post the output of the Diagnostics report even if 100% successful just in case some target has unexpected performance issues.

Platforms of particular interest (reports from others are appreciated)

- [ ] Raspberry Pi Zero W
- [ ] Raspberry Pi Zero 2W
- [ ] Raspberry Pi 3
- [ ] Raspberry Pi 4
- [ ] Raspberry Pi 5
- [ ] BeagleBoneBlack or green, etc. nerves_system_bbb 2.20.1
- [ ] BeagleBoneBlack or green, etc. nerves_system_bbb 2.19.x or earlier
- [x] MangoPi MQ-Pro
- [ ] GRiSP

fhunleth, 2023-12-30:
Here's the MangoPi:

```
iex(1)> Circuits.GPIO2.Diagnostics.report(36, 35)
Circuits.GPIO Diagnostics

GPIO 1: 36
GPIO 2: 35
Backend: Circuits.GPIO2.CDev

Writes can be read 1->2: PASSED
Writes can be read 2->1: PASSED
Can set 0 on open: PASSED
Can set 1 on open: PASSED
Input interrupts sent: PASSED
Interrupt timing sane: PASSED
Internal pullup works: PASSED
Internal pulldown works: PASSED

Speed test: 47853 toggles/second

All tests passed
```

fhunleth, 2023-12-30:
Here's the result on a BeagleBone Green Wireless:

```
iex(4)> Circuits.GPIO.Diagnostics.report(60, 112)
Circuits.GPIO Diagnostics 2.0.0-pre.2

Output GPIO: 60
Input GPIO: 112
Backend: Circuits.GPIO.CDev

Simple writes and reads work: PASSED
Can set 0 on open: PASSED
Can set 1 on open: PASSED
Input interrupts sent: PASSED
Interrupt timing sane: PASSED
Internal pullup works: PASSED
Internal pulldown works: FAILED 291: Assertion failed: GPIO.read(in_gpio) == 0

Speed test: 69713 writes/second

Failed
```

fhunleth, 2023-12-31:
Raspberry Pi 5
(I had to specify pins by label rather than by GPIO number. The pins were on gpiochip4)

```
iex(21)> Circuits.GPIO.Diagnostics.report("PIN36", "PIN38")
Circuits.GPIO Diagnostics 2.0.0-pre.2

Output GPIO: "PIN36"
Input GPIO: "PIN38"
Backend: Circuits.GPIO.CDev

Simple writes and reads work: PASSED
Can set 0 on open: PASSED
Can set 1 on open: PASSED
Input interrupts sent: PASSED
Interrupt timing sane: PASSED
Internal pullup works: FAILED 268: Assertion failed: GPIO.read(in_gpio) == 1
Internal pulldown works: FAILED 291: Assertion failed: GPIO.read(in_gpio) == 0

Speed test: 646412 writes/second

Failed
```

pojiro, 2024-01-01:
I tested Raspberry Pi 4 with `2.0.0-pre.2`.

```elixir
iex(2)> Circuits.GPIO.Diagnostics.report(16, 20)
Circuits.GPIO Diagnostics 2.0.0-pre.2

Output GPIO: 16
Input GPIO: 20
Backend: Circuits.GPIO.CDev

Simple writes and reads work: FAILED 173: Assertion failed: GPIO.read(in_gpio) == 1
Can set 0 on open: PASSED
Can set 1 on open: FAILED 189: Assertion failed: GPIO.read(in_gpio) == value
Input interrupts sent: FAILED Expected message not received within 500ms: {:circuits_gpio, ^in_gpio_spec, _timestamp, 1}
Interrupt timing sane: FAILED Expected message not received within 500ms: {:circuits_gpio, ^in_gpio_spec, _, 1}
Internal pullup works: FAILED 258: Assertion failed: GPIO.read(in_gpio) == 0
Internal pulldown works: FAILED 281: Assertion failed: GPIO.read(in_gpio) == 1

Speed test: 348250 writes/second

Failed

```

pojiro, 2024-01-01:
I tested Raspberry Pi 3 with `2.0.0-pre.2`.

```elixir
iex(1)> Circuits.GPIO.Diagnostics.report(16, 20)
Circuits.GPIO Diagnostics 2.0.0-pre.2

Output GPIO: 16
Input GPIO: 20
Backend: Circuits.GPIO.CDev

Simple writes and reads work: FAILED 173: Assertion failed: GPIO.read(in_gpio) == 1
Can set 0 on open: PASSED
Can set 1 on open: FAILED 189: Assertion failed: GPIO.read(in_gpio) == value
Input interrupts sent: FAILED Expected message not received within 500ms: {:circuits_gpio, ^in_gpio_spec, _timestamp, 1}
Interrupt timing sane: FAILED Expected message not received within 500ms: {:circuits_gpio, ^in_gpio_spec, _, 1}
Internal pullup works: FAILED 258: Assertion failed: GPIO.read(in_gpio) == 0
Internal pulldown works: FAILED 281: Assertion failed: GPIO.read(in_gpio) == 1

Speed test: 238550 writes/second

Failed
```

fhunleth, 2024-01-01:
@pojiro Could you confirm that GPIOs 16 and 20 are connected together? The results that you're getting suggest that there's no wire connecting them.

fhunleth, 2024-01-01:
GRiSP2 with the GPIO1's PMOD connector pins 1 and 3 connected. (pin 2 didn't work?)

```
iex(26)> Circuits.GPIO.Diagnostics.report({"gpiochip0", 1}, {"gpiochip0", 3})
Unexpected return from reading gpio events: -1, errno=9
Circuits.GPIO Diagnostics 2.0.0-pre.2

Output GPIO: {"gpiochip0", 1}
Input GPIO: {"gpiochip0", 3}
Backend: Circuits.GPIO.CDev

Simple writes and reads work: PASSED
Can set 0 on open: PASSED
Can set 1 on open: PASSED
Input interrupts sent: PASSED
Interrupt timing sane: PASSED
Internal pullup works: FAILED 268: Assertion failed: GPIO.read(in_gpio) == 1
Internal pulldown works: FAILED 291: Assertion failed: GPIO.read(in_gpio) == 0

Speed test: 46848 writes/second
```

pojiro, 2024-01-01:
@fhunleth I didn't understand the protocol. Now I see it, tested again. So I updated the results. Thanks.

pojiro, 2024-01-01:
I tested BeagleBoneBlack nerves_system_bbb 2.19.1 with `2.0.0-pre.2`.

```elixir
iex(2)> Circuits.GPIO.Diagnostics.report(67, 68)
Circuits.GPIO Diagnostics 2.0.0-pre.2

Output GPIO: 67
Input GPIO: 68
Backend: Circuits.GPIO.CDev

Simple writes and reads work: PASSED
Can set 0 on open: PASSED
Can set 1 on open: PASSED
Input interrupts sent: PASSED
Interrupt timing sane: PASSED
Internal pullup works: PASSED
Internal pulldown works: FAILED 291: Assertion failed: GPIO.read(in_gpio) == 0

Speed test: 76327 writes/second

Failed
```

fhunleth, 2024-01-01:
@pojiro Great! Thank you for retesting and updating the reports. I'll be updating the diagnostics for the `pre.3` release to make this more clear and also to give more explicit feedback when the basic test fails.

fhunleth, 2024-01-02:
`circuits_gpio v2.0.0-pre.3` is available now. It doesn't add any new checks to the diagnostics, but it does have more help when tests fail.

Moosieus, 2024-01-02:
Raspberry Pi Zero W on `2.0.0-pre.3`:
```
iex(4)> Circuits.GPIO.Diagnostics.report(16, 20)

Circuits.GPIO Diagnostics 2.0.0-pre.3

Output GPIO: 16
Input GPIO:  20

Output info: %Circuits.GPIO.Line{gpio_spec: {"gpiochip0", 16}, label: "GPIO16", controller: "pinctrl-bcm2835"}
Input info:  %Circuits.GPIO.Line{gpio_spec: {"gpiochip0", 20}, label: "GPIO20", controller: "pinctrl-bcm2835"}
Backend: Circuits.GPIO.CDev

Simple writes and reads work: PASSED
Can set 0 on open: PASSED
Can set 1 on open: PASSED
Input interrupts sent: PASSED
Interrupt timing sane: PASSED
Internal pullup works: PASSED
Internal pulldown works: PASSED

Writes/second: 117302
Reads/second: 118245

All tests passed
true
```

fhunleth, 2024-01-03:
@Moosieus Thanks for posting that. I just now noticed that the labels for pins 16 and 20 are "GPIO16" and "GPIO20" for that Raspberry Pi Zero, and that's different from the Raspberry Pi 5 (PINn where n is the physical GPIO connector pin number). That will certainly be confusing to people who don't know, so I added a note to the docs.

Moosieus, 2024-01-03:
Here's the Raspberry Pi Zero 2W on `2.0.0-pre.3` as well:
```
iex(1)> Circuits.GPIO.Diagnostics.report(16, 20)

Circuits.GPIO Diagnostics 2.0.0-pre.3

Output GPIO: 16
Input GPIO:  20

Output info: %Circuits.GPIO.Line{gpio_spec: {"gpiochip0", 16}, label: "GPIO16", controller: "pinctrl-bcm2835"}
Input info:  %Circuits.GPIO.Line{gpio_spec: {"gpiochip0", 20}, label: "GPIO20", controller: "pinctrl-bcm2835"}
Backend: Circuits.GPIO.CDev

Simple writes and reads work: PASSED
Can set 0 on open: PASSED
Can set 1 on open: PASSED
Input interrupts sent: PASSED
Interrupt timing sane: PASSED
Internal pullup works: PASSED
Internal pulldown works: PASSED

Writes/second: 230044
Reads/second: 232477

All tests passed
true
```

fhunleth, 2024-01-03:
All boards are covered! Thanks @pojiro and @Moosieus!

I'm going to leave this issue open a little longer to review the failures. I think the failures are expected due to hardware limitations, though.

fhunleth, 2024-01-10:
Closing since the `v2.0` updates are looking pretty good on official systems and proprietary boards and software.

closed 2024-01-10 by fhunleth