elixir-circuits/circuits_gpio #191: `backend_info/0` reports `Circuits.GPIO.CDev` instead of `:stub` as name

namxam, 2024-04-03:
It seems that with the 2.0 release, the old `info/0` has been replaced with `backend_info/0`. At the same time, the function now returns `Circuits.GPIO.CDev` as the current backend name instead of `:stub`. Is this intended? Because the docs are still stating that it should be `:stub`. See https://hexdocs.pm/circuits_gpio/readme.html#testing

fhunleth, 2024-04-03:
@namxam Thank you for point this out. This was not intentional at all! I totally missed that I hadn't updated that part and forgot to remove the hack that overrode the `:name` field when it wasn't ready.

I fixed it in #192.

Now the `:name` field will properly reflect the way the backend was started. (In hindsight, I wish I chose a different word than `:name`, but the change in #192 keeps non-test usage of `backend_info/0` the same.)

```
$ MIX_ENV=test iex -S mix
"**** CIRCUITS_GPIO_BACKEND set to [test] ****"
Erlang/OTP 26 [erts-14.2.3] [source] [64-bit] [smp:32:32] [ds:32:32:10] [async-threads:1] [jit:ns]

Interactive Elixir (1.16.2) - press Ctrl+C to exit (type h() ENTER for help)
iex(1)> Circuits.GPIO.backend_info()
%{name: {Circuits.GPIO.CDev, [test: true]}, pins_open: 0}
iex(2)> 
```

fhunleth, 2024-04-08:
Fixed in #192